sourceperl/pyModbusTCP #7: last_error() and last_except()

suibaf, 2017-01-10:
Can you give an example on how to use this two function please?
Best regards

sourceperl, 2017-01-12:
```python

#!/usr/bin/env python
# -*- coding: utf-8 -*-

import time
from pyModbusTCP.client import ModbusClient
import pyModbusTCP.constants as const

c = ModbusClient(host="192.168.0.100", auto_open=True)

while True:
    # read 10 bits at address 20480
    bits = c.read_coils(20480, 10)
    # if read ok
    if bits:
        print("bit ad #0 to 9: "+str(bits))
    # if read error
    else:
        # check modbus exception
        if c.last_error() == const.MB_EXCEPT_ERR:
            print("read return exception (except code=%i)" % c.last_except())
        else:
            print("read error")

    # sleep 2s
    time.sleep(2)
```

closed 2017-03-28 by sourceperl

GrantEdwards, 2021-02-25:
Are there any functions to produce something human readable for an error code or an exception code?

sourceperl, 2021-02-26:
I've just add this, you can use last_error_txt() and last_except_txt(). You need to upgrade to v0.1.9 before.