sourceperl/pyModbusTCP #9: AttributeError: 'module' object has no attribute 'inet_pton'

rolliracker, 2017-03-27:
Hi,
First, thank you very much for the code!
On Windows 7, using Python 2.7 (32 bit), when I tried 
`c = ModbusClient(host="127.0.0.1", auto_open=True, auto_close=True)`
I got an exception saying:
`Traceback (most recent call last):
  File "<pyshell#5>", line 1, in <module>
    c = ModbusClient(host="127.0.0.1", auto_open=True, auto_close=True)
  File "C:\WinPython-32bit-2.7.10.3\python-2.7.10\lib\site-packages\pyModbusTCP\client.py", line 61, in __init__
    if not self.host(host):
  File "C:\WinPython-32bit-2.7.10.3\python-2.7.10\lib\site-packages\pyModbusTCP\client.py", line 126, in host
    socket.inet_pton(socket.AF_INET, hostname)
AttributeError: 'module' object has no attribute 'inet_pton'
`
I solved this by running `pip install win-inet-pton` and adding `import win_inet_pton` in my code, as explained here: [https://github.com/mitsuhiko/python-geoip/issues/4](url) 

Roland

dbe-hms, 2017-09-29:
Thanks I was running into this same problem!

muthunsr, 2018-09-26:
Thanks i had same issue. Now the code is running successfully!!

shachashou, 2018-10-09:
Thanks , it  helps me a lot

sourceperl, 2020-05-05:
I add notice to README to warn windows user about this.

closed 2020-05-05 by sourceperl