sourceperl/pyModbusTCP #10: Passing a float value to write_single_register

gcezar, 2017-10-30:
Is there a way to pass a float value to write_single_register function? My device requires a float number to be passed and it looks like write_single_register only accepts int.

Thanks

sourceperl, 2017-10-30:
You can't use write_single_register for this because this function write 16 bits wide word only and float require 32 bits wide so 2 words.

Try this:

`from pyModbusTCP.client import ModbusClient
from pyModbusTCP import utils

c = ModbusClient(host="localhost", port=502, auto_open=True)
# convert python float 0.3 to int 0x3e99999a (32 bits IEEE representation)
val_as_int = utils.encode_ieee(0.3)
# write 2 words at address 0
c.write_multiple_registers(0, [(val_as_int&0xffff0000)>>16, val_as_int&0xffff])`

gcezar, 2017-11-06:
Great, that worked!

Thanks!

closed 2017-11-06 by gcezar