sourceperl/pyModbusTCP #12: Bug with multiple TCP packets.

Farin94, 2018-05-11:
Hi there

If found a bug that gives a wrong error when a modbus packet is split up in multiple tcp packets. The recv function only reads the buffer for the first packet. I propose the following fix in the _recv function of the client.

`    def _recv(self, max_size):
        """Receive data over current socket

        :param max_size: number of bytes to receive
        :type max_size: int
        :returns: receive data or None if error
        :rtype: str (Python2) or class bytes (Python3) or None
        """
        # wait for read
        if not self._can_read():
            self.close()
            return None
        # recv
        try:
            ### Here's the modified receiver code
            r_buffer = bytearray()
            while len(r_buffer) < max_size:
                r_buffer += self.__sock.recv(max_size - len(r_buffer))
        except socket.error:
            print(socket.error)
            r_buffer = None
        # handle recv error
        if not r_buffer:
            self.__last_error = const.MB_RECV_ERR
            self.__debug_msg('_recv error')
            self.close()
            return None
        return r_buffer`

Would it be possible to include this?

Thanks alot

Farin

sourceperl, 2018-05-11:
Hi,

You encounter this behavior with modbus TCP or RTU ?

Farin94, 2018-05-11:
This was in modbusTCP.

sourceperl, 2018-05-14:
Can you test your code with the last commit 9b05da9f8bb9febcebf2bd0c93ccbbcdf816093f ? This one include the "_recv_all" function to deal with multiple packets when use in modbus TCP mode.

I have no device with the fragment behaviour so it's difficult to easy test this for me...

Farin94, 2018-05-14:
Works like a charm! Thank you very much!

sourceperl, 2018-05-14:
Thanks for the report !

closed 2018-05-14 by sourceperl