sourceperl/pyModbusTCP #18: Proper way to end server instance running inside a thread?

Rugiewitz, 2018-10-12:
So I create a function that creates the ModusServer Object and starts it,
I have a thread that assigned that function. 
So for a thread to end normally it needs to wait for the completion of the function, but the problem is that the function is never gonna end because the ModbusServer is always running.

My question is, is there a way to make that server object end by a function call so than i'm able to properly kill/terminate/end my thread

sourceperl, 2018-10-15:
Hi,

I've just add stop() and is_run property to ModbusServer. This permit to control the thread when use ModbusServer with no_block option set.

Now, we can control the thread like this:
```python
import argparse
import time
from pyModbusTCP.server import ModbusServer

if __name__ == '__main__':
    # parse args
    parser = argparse.ArgumentParser()
    parser.add_argument('-H', '--host', type=str, default='localhost', help='Host')
    parser.add_argument('-p', '--port', type=int, default=502, help='TCP port')
    args = parser.parse_args()

    # init modbus server
    server = ModbusServer(host=args.host, port=args.port, no_block=True)
    # start with no_block option set (start server in another thread)
    server.start()
    # check running
    if server.is_run:
        print("server is run")
    # wait 10s
    time.sleep(10)
    # stop the server thread and close the socket
    server.stop()
    # check running
    if not server.is_run:
        print("server is stop")
```

Rugiewitz, 2018-10-31:
Thank you so much!!

closed 2018-10-31 by Rugiewitz