sourceperl/pyModbusTCP #21: How to set Device ID ?

mridah, 2019-04-15:
This is my code:

    c = ModbusClient(host="localhost", port=502, auto_open=True)
    regs = c.read_holding_registers(0, 2) # i want to mention slave ID here
    if regs:
        print(regs)
    else:
        print("read error")

How do I set a slave ID/device ID to specify a particular device I want to query ??

RottenVanSuti, 2019-04-16:
@mridah  Check the link with documentation in section PyModbus Module TCP client, the constructor has the ID parameter

amarpremk, 2019-11-03:
new to this @mridah please share your example prog if you are able to add slave ID.

mridah, 2019-11-03:
@amarpremk you can use `unit` for specifying the slave ID

The following code will return the first holding register ( 40001 ) of device with slave ID 10

    modbusClient = ModbusClient(host="localhost", port=502, auto_open=True)
    value = modbusClient.read_holding_registers(address=0, count=1, unit=10)

sourceperl, 2020-05-05:
I update README (222677acf539c8cff4ce57e946ccccdcc0d94f6e) with unit_id (new name of slave id in modbus/TCP) mention to avoid misunderstood.

closed 2020-05-05 by sourceperl