sourceperl/pyModbusTCP #22: Strange float value reading

RottenVanSuti, 2019-04-18:
Hi @sourceperl I've been testing some Holding register readings with uint and int values from a energy meter and is ok, but there some values in float format. When I try to read the float values, the result is a very small number (for example, 3.44533e-35) and does not match the real value that's shown in the energy meter (let's say it shows 35.23 kW). I've used as guide the TCP float example. I don't know if this problem is related to the equipment
I attached a image with the modbus addresses. I use addresses from 350 to 355
[image]

sourceperl, 2019-04-23:
Hi, your document say data type is long for address 350 to 355. So, it's probably not float but 32 bits signed int. To deal with 32 bits you can use utils module ([online doc](https://pymodbustcp.readthedocs.io/en/latest/quickstart/index.html#utils-module-modbus-data-mangling)).

RottenVanSuti, 2019-04-28:
Thank you @sourceperl , I'm sorry, you're right, I got confused Long variable with Float, now I get the correct vaues. However, there some negative values in the energy meter that seems they can not be handle with the library, because when I print the result, it shows the max value of the 16bit register, no the negative value

RottenVanSuti, 2019-04-28:
I'm sorry, I've been reading about the" two complement" utils. This is the function that I should use for the Long (signed) values, right?

sourceperl, 2019-04-28:
You can use [pyModbusTCP.utils.get_2comp](https://pymodbustcp.readthedocs.io/en/latest/package/module_utils.html#pyModbusTCP.utils.get_2comp) for deal with 2's complement integer. In your case, since your registers are long, you have to set val_size parameter to 32.

Like this:
```python
from pyModbusTCP import utils

# 2's complement decode of 0xFFFFFFFF will give you -1
print(utils.get_2comp(0xFFFFFFFF, val_size=32))
```

RottenVanSuti, 2019-04-30:
Thank you @sourceperl , it worked

sourceperl, 2019-05-06:
I'm glad to hear that !

closed 2019-05-06 by sourceperl