sourceperl/pyModbusTCP #23: timeout error _recv MBAP error

polubencev, 2019-05-15:
LinuxMint18. Error timeout error
_recv MBAP error. Port 502 is close? Help..

closed 2021-02-26 by sourceperl

desmondsam, 2022-03-11:
Hi, I am seeing the same: 
timeout error 
_recv MBAP error. 
All I am trying to do is read an input register. I have my code below. Any help is much appreciated.

from pyModbusTCP.client import ModbusClient
c = ModbusClient('192.168.255.101', debug=True, timeout=30)
c.open()
True
c.read_input_registers(12)
Tx
[33 EB 00 00 00 06 01] 04 00 0C 00 01
timeout error
_recv MBAP error

sourceperl, 2022-03-14:
Hi, hard to say what's wrong here. If you can see the tx frame in debug mode, the TCP socket is open and ready for data exchange. Here it is the server that did not return anything. If you are using a modbus gateway, maybe the unit id does not match the slave address ?

desmondsam, 2022-03-14:
Hi,

Thank you for your response. I looked up the unit id on the slave and added it and I still get the same error. Pasted the console output here. New to the whole Modbus thing, please let me know if you have any suggestions. I will update here if I find out a fix. Thank you!

>>> from pyModbusTCP.client import ModbusClient
>>> c = ModbusClient('192.168.255.101', debug=True, timeout=30, unit_id=2, port=502)
>>> c.open()
True
>>> c.read_input_registers(12)
Tx
[D3 BB 00 00 00 06 02] 04 00 0C 00 01
timeout error
_recv MBAP error

desmondsam, 2022-03-14:
Hi,

I am not sure if this is the fix, but disabling firewall on my PC and on the device I am trying to communicate to from the PC, I had to set the default gateway to 192.168.255.1 and this fixed the timeout error. Thank you!