sourceperl/pyModbusTCP #24: How to read slave ID

hsoydam, 2019-09-08:
I started writing a new application in Python.
I am trying to read using slave id with modbustcp.
Can you share the reading instance with me slave id?

amarpremk, 2019-11-02:
Same here..!!
An example with slave ID would be very much useful ..

sourceperl, 2020-05-05:
I update basic example read_bit.py with unit_id (modbus/TCP new name for slave id) choice (link to update: aa069af378447391fe0511ad66ff063731f055b4).

Also, you can add unit_id directly to ModbusClient constructor like this:
`
c = ModbusClient(host='127.0.0.1', unit_id=1)
`

sourceperl, 2020-05-05:
I update README (222677acf539c8cff4ce57e946ccccdcc0d94f6e) with unit_id (new name of slave id in modbus/TCP) mention to avoid misunderstood.

closed 2020-05-05 by sourceperl