sourceperl/pyModbusTCP #28: Increase the number of registers read in one request [REQUEST]

castillo92, 2020-04-29:
I have a Fronius Solar inverter that is a bit slow answering throught ModbusTCP because of the request. But only in the request.

Could be increased the number of registers in one request from this libray? Now, according to the docs, the maximun number of registers to read is 125.

[Link to docs](https://pymodbustcp.readthedocs.io/en/latest/package/class_ModbusClient.html#pyModbusTCP.client.ModbusClient.read_holding_registers)

**read_input_registers(reg_addr, reg_nb=1)**
Parameters:
- reg_addr (int) – register address (0 to 65535)
- reg_nb (int) – number of registers to read (1 to 125)

Could be increased at 250?

sourceperl, 2020-05-05:
It's a protocol limitation introduce in Modbus/TCP for compatibility with Modbus RTU (256 bytes max/frame). So it can't be increase.

closed 2020-05-05 by sourceperl

castillo92, 2020-05-05:
Thanks you very much.