sourceperl/pyModbusTCP #29: Reading holding registers with negative values

briantilburgs, 2020-06-04:
Hi,

Today I hava installed a EMU Modbus TCP power meter for my solat energy.
I thried to measure the returned energy but thet seems to be [65535, 65432] I think that should be a negative value, how should I correct that?

emu = ModbusClient(host=emu_dev["ip"], port=emu_dev["port"], auto_open=True)
emu.read_holding_registers(4524, 4)

Thanks,

Brian

sourceperl, 2020-06-05:
Hi, yes it's probably "two’s complement" negative value. 

See at https://pymodbustcp.readthedocs.io/en/latest/quickstart/index.html#utils-module-modbus-data-mangling for how to deal with this.

solanu, 2020-07-24:
Same here but writing negative int32 (long).

I have transformed negative int (-250) to word list using `long_list_to_word` function and I get [65286, -1] when it should be [65286, 65535]

sourceperl, 2021-02-27:
Not sure if this is still useful for you, but the behavior you report in july matches the "historical" use of get_2comp. That is to say, he could translate data received via modbus and convert them. But it can not do it in the other way with negative python int.

Quickstart extract with historical usage:
```python
from pyModbusTCP import utils
list_16_bits = [0x0000, 0xFFFF, 0x00FF, 0x8001]

# display "[0, -1, 255, -32767]"
print(utils.get_list_2comp(list_16_bits, 16))

# display "-1"
print(utils.get_2comp(list_16_bits[1], 16))
```
Since commit 644ed80359b2166f377cfe691c626cc5773709f0  get_2comp deal with negative python int also.

So now:
```python
from pyModbusTCP.utils import get_2comp, long_list_to_word

# return [65286, 65535]
print(long_list_to_word([get_2comp(-250, 32)], big_endian=False))
```

closed 2021-12-22 by sourceperl