sourceperl/pyModbusTCP #33: How to write to a holding register properly?

Boulder08, 2020-10-13:
I'm sorry to open this as an issue, but I cannot get my head around this problem as a Modbus newbie. I'm trying to control my heating system by adjusting the temperature setting for water in various outdoor temperature ranges. I've been able to find the holding registers at 502, 506, 510, 514 and 518. However, I don't understand how I can write to those positions. I've tested

`c.write_single_register(510, 20)`
but it doesn't change the value.

This is how I've found the values:
```
from pyModbusTCP.client import ModbusClient
c = ModbusClient(host="192.168.0.11", port=8899, auto_open=True, auto_close=True)
print (c.read_holding_registers(510, 1))
```

This could be something to add to the examples, perhaps?

sourceperl, 2021-02-27:
I'm not sure if you still have the problem since October. But, just in case, you can try enabling ModbusClient debug mode and see what happens.

```python
from pyModbusTCP.client import ModbusClient

# set debug to True will echo the tx/rx frame, it's more easy to see what happen
c = ModbusClient(host="192.168.0.11", port=8899, auto_open=True, auto_close=True, debug=True)
print (c.read_holding_registers(510, 1))
```

Boulder08, 2021-03-12:
This was actually a quite simple issue, I just didn't understand that I needed to use write_multiple_registers to make it work. Probably due to those registers being 32bit? Also the specifications have a different offset from pyModbusTCP so it threw me another curveball :)

For example this is what I've used to write a temperature value in my home automation script:
modbus.write_multiple_registers(509, [0, 40])

sourceperl, 2021-03-12:
Nice to see it's ok for you.

closed 2021-03-12 by sourceperl