sourceperl/pyModbusTCP #36: Not able to add slave ID in holding register function

vidyarao123, 2021-02-01:
Hi , 

I am trying to add the slave id (or unit) in the read_holding_registers part 
As per the docs, the syntax is :- read_holding_registers(address,count,unit) but 

when I am trying to follow the same in my code (as shown below)
**HR = client.read_holding_registers(Reg_Address,100,1)** (This line is from my python code)

I am getting an error like this 
TypeError: read_holding_registers() takes from 2 to 3 positional arguments but 4 were given

Is there a mistake in the format that I have given. Can somebody pls check this?

sourceperl, 2021-02-01:
Hi, i'm not sure to understand. There are no unit_id arg in "read_holding_registers". You have to set it with ModbusClient unit_id(MY_ID) method or directly in ModbusClient constructor.

https://pymodbustcp.readthedocs.io/en/latest/package/class_ModbusClient.html#pyModbusTCP.client.ModbusClient.read_holding_registers

Hope this  help you.

vidyarao123, 2021-02-17:
Yes, Its working. Document was helpful 

Thanks

closed 2021-02-17 by sourceperl