sourceperl/pyModbusTCP #37: Data change notification in pymodbusTCP

vidyarao123, 2021-03-03:
Hi,

Is it possible to get a notification on pymodbusTCP client whenever there is a change in the value in Modbus Server ? 

If there is no inbuilt functionality as such, can somebody explain how to achieve this ?

sourceperl, 2021-03-12:
Hi,

It's not easy to do this now since the data requests from the Modbus server are directly linked to DataBank. 

I'll working to improve this, probably with the addition of some abstraction between the server and the data.

sourceperl, 2022-06-05:
I'm not sure this is what you originally wanted, but from now (release 0.2.0) the server can notify a change of values in its address space (see examples/server_change_log.py).

closed 2022-06-05 by sourceperl