sourceperl/pyModbusTCP #40: server.start() results in error 13 on Ubuntu

ericboxer, 2021-05-25:
I've been working on getting this module integrated into a project and seems to run fine on my Mac. As soon as I move it over to a staging platform (Ubuntu) I get an error 13 as soon as server.start() is run.

I have given full permissions to all of my files, as well as trying to put the pyModbusTCP files in the same directory with full permissions. 

Any thoughts on where I can look next?

sourceperl, 2021-05-25:
You must start the script as root to open a TCP port <1024 (as for the default modbus port 502).

more info : https://stackoverflow.com/questions/24001147/python-bind-socket-error-errno-13-permission-denied

ericboxer, 2021-06-01:
Ahhh that makes sense, thank you!

closed 2021-06-01 by ericboxer