sourceperl/pyModbusTCP #41: Program hangs when not using no_block=true

abishur, 2021-06-16:
Using a Raspberry Pi Zero I installed pip3 and then install pyModbusTCP using pip3

Using the default example for a server I placed a print("Started") after server.start(), but it never prints started.  If I added no_block=true then it will print "started" but in both cases I am unable to connect to the server.

Any ideas what could be going on?

abishur, 2021-06-16:
I have also tried using the library on Ubuntu on a regular computer, and I cannot connect to the modbusTCP server either locally or remotely.  I've tried installing the past 2 versions and those didn't work either.

abishur, 2021-06-17:
After doing some testing I've discovered that if I put the actual IP address of the computer I can connect to it, but if I give localhost or 127.0.0.1 then the connection fails.  For the time being I've added this bit of code to fix the issue:

`
from pyModbusTCP.server import ModbusServer, DataBank
import socket

s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
try:
  # doesn't even have to be reachable
  s.connect(('10.255.255.255', 1))
  IP_add = s.getsockname()[0]
except Exception:
  IP = '127.0.0.1'
finally:
  s.close()

server = ModbusServer(host=IP_add,port=502,no_block=True)
server.start()
`

And as long as I run the script with root privileges (though it would also work without root privileges if I chose a higher port number) I can successfully connect to it.  Alternatively, I could explicitly state the IP address and not need to import socket, but if the computer is getting it's ip through DHCP anytime the address changes the script would break.

sourceperl, 2021-06-17:
You should try to use the ModbusServer class with the host parameter set to '0.0.0.0' (= listen on all host IPs).

You can also look at https://stackoverflow.com/questions/38256851/why-does-my-python-tcp-server-need-to-bind-to-0-0-0-0-and-not-localhost-or-its

abishur, 2021-06-17:
Thank you, setting it to 0.0.0.0 did indeed allow me to connect to it (and without needing to import socket), you might want to update the example sketch to make a note of this as the documentation and all examples show 'localhost' or 127.0.0.1

sourceperl, 2021-06-18:
You're right, I'll add a few comments.

closed 2021-06-18 by sourceperl