sourceperl/pyModbusTCP #50: Module not found error

IGDT, 2022-03-03:
On my raspberry pi 4 I tried to code some basic stuff (like printing messages) to test out the pymodbustcp.
For some reason, it doesn't want to find the module of pymodbustcp and it gives this message when I run the code:
`/home/pi/Desktop/Laadpunt/venv/bin/python /home/pi/Desktop/Laadpunt/main.py
Traceback (most recent call last):
  File "/home/pi/Desktop/Laadpunt/main.py", line 7, in <module>
    from pyModbusTCP.client import ModbusClient
ModuleNotFoundError: No module named 'pyModbusTCP'

Process finished with exit code 1
`
I installed both "sudo pip2 install pyModbusTCP" as "sudo pip3 install pyModbusTCP" in my console and that says:
`pi@raspberrypi:~ $ sudo pip install pyModbusTCP
Traceback (most recent call last):
  File "/usr/local/bin/pip", line 6, in <module>
    from pip._internal.cli.main import main
ModuleNotFoundError: No module named 'pip._internal.cli.main'
pi@raspberrypi:~ $ 
`
What can I do to fix this issue?

superpanza, 2022-03-04:
Did you install the library after the venv has been activated? It seems that you install the library system wide and then launch the python executable inside the virtual env. Try to activate the virtual env first with `source /home/pi/Desktop/Laadpunt/venv/bin/activate`. Your terminal should prompt something like  `(venv) pi@raspberrypi:`  then install the lib via pip `pip install pyModbusTCP`. Now you can launch your code with` python  /home/pi/Desktop/Laadpunt/main.py` or move inside the folder and ` python main.py`. Hope this can help

IGDT, 2022-03-04:
> Did you install the library after the venv has been activated? It seems that you install the library system wide and then launch the python executable inside the virtual env. Try to activate the [...]

This worked, thank you for the response. Do all modules have to be installed like this? (so I don't mess up in the future)

sourceperl, 2022-06-05:
"Do all modules have to be installed like this? (so I don't mess up in the future)"

Yes, if you are in a virtual environment. It's a good practice, which I don't use often enough... 😉

closed 2022-06-05 by sourceperl