sourceperl/pyModbusTCP #52: Error: TypeError: object.__init__() takes exactly one argument (the instance to initialize)

ksaye, 2022-03-24:
Any suggestion how to address this error?

```
ksaye@home:~$ wget https://raw.githubusercontent.com/sourceperl/pyModbusTCP/master/examples/server_virtual_data.py
--2022-03-24 11:36:53--  https://raw.githubusercontent.com/sourceperl/pyModbusTCP/master/examples/server_virtual_data.py
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.111.133, 185.199.108.133, 185.199.109.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1870 (1.8K) [text/plain]
Saving to: ‘server_virtual_data.py’

server_virtual_data.py                                                          100%[====================================================================================================================================================================================================>]   1.83K  --.-KB/s    in 0s

2022-03-24 11:36:53 (25.2 MB/s) - ‘server_virtual_data.py’ saved [1870/1870]

ksaye@home:~$ ls -all server_virtual_data.py
-rw-rw-r-- 1 ksaye ksaye 1870 Mar 24 11:36 server_virtual_data.py

ksaye@home:~$ python3 server_virtual_data.py
Traceback (most recent call last):
  File "server_virtual_data.py", line 48, in <module>
    server = ModbusServer(host=args.host, port=args.port, data_bank=MyDataBank())
  File "server_virtual_data.py", line 25, in __init__
    super().__init__(virtual_mode=True)
TypeError: object.__init__() takes exactly one argument (the instance to initialize)
ksaye@home:~$
```

sgrund, 2022-04-06:
Same problem for me.
Did you find a solution?

sourceperl, 2022-06-05:
Probably because at the time you write your post, the version on the github repository and the examples too, correspond to the development version 0.2.0-dev. On the other hand, the stable version, the one available on pypi, is 0.1.10. Hence the problem when running 'server_virtual_data.py'.

Today I am promoting 0.2.0-dev to 0.2.0 stable. So, at the moment, pypi and the repository have the same version 0.2.0.

To solve your problem, you can do:
```bash
sudo pip3 install pyModbusTCP --upgrade
```

closed 2022-06-05 by sourceperl