sourceperl/pyModbusTCP #54: Modbus function code 2 not supported?

ravindragurjar, 2022-05-18:
pyModbusTCP package present not supporting modbus function code 2 (read input status)

Is there any other way to handle this in current package?

sourceperl, 2022-06-05:
Modbus function code 2 is available in ModbusClient, like this:

```python
from pyModbusTCP.client import ModbusClient

# init modbus client
c = ModbusClient(host='myservername')

# do function 2 read at @0
d_inputs_l = c.read_discrete_inputs(0)
print(d_inputs_l)
```

It's also available in modbus server:

```python
import time
from pyModbusTCP.server import ModbusServer

# init modbus server
server = ModbusServer(host='localhost', no_block=True)
server.start()

b_flag = False

while True:
    b_flag = not b_flag
    server.data_bank.set_discrete_inputs(0,  [b_flag])
    time.sleep(5.0)
```

annishao, 2022-08-29:
I agree, I am experiencing some issues with function code 2. It only returns values for certain ranges, but otherwise it will return error code 130 (hex 0x82). Let me know if you need further elaboration.