sourceperl/pyModbusTCP #60: Device Identification objects

dunforda, 2022-09-06:
Thank you very much for your package.
Is it possible to read the Device Identification information?
E.g. serial number of TH probes etc.

Thank you,

dunforda, 2022-09-06:
I tried copy/adapting read_holding_registers in client.py but it's obviously a bit more complicated than that ...

sourceperl, 2023-11-21:
This is now available in pyModbusTCP version 0.2.1.

```python
from pyModbusTCP.client import ModbusClient

mc = ModbusClient(host='myserver.example.com')
dev_id_resp = mc.read_device_identification()
print(f'vendor={dev_id_resp.vendor_name}')
```

more: https://pymodbustcp.readthedocs.io/en/latest/package/class_ModbusClient.html#pyModbusTCP.client.ModbusClient.read_device_identification

closed 2024-07-30 by sourceperl