sourceperl/pyModbusTCP #64: ModbusServer unit_id?

mgillard-sudo, 2022-12-28:
Just wondering if it's possible to start the server with multiple slaves/unit_ids as I'm attempting to make a device simulator so being able to have a host a server, with multiple slaves that can be written to would be useful. If this is possible please let me know as I've been looking through the documentation for a day or so and have no clue.

Many thanks,

Max.

charliehuddleston, 2022-12-28:
Looking through the server source it looks like the unit_id is not currently used for anything.  It looks like the unit_id gets packed into the mbap, so depending on what operations you are expecting from the clients you could override them and manually map to different databanks depending on the unit_id.  But, if you have the ability to use different ports on your network in addition to just 502, it might be easier to instantiate multiple servers on different ports, or have register offsets for different simulated devices (i.e. device 1 is registers 1-999, device 2 is 1001-1999 etc).

mgillard-sudo, 2022-12-29:
> Looking through the server source it looks like the unit_id is not currently used for anything.  It looks like the unit_id gets packed into the mbap, so depending on what operations you are [...]

Thanks I'll probably just use additional ports, I was hoping there was a simple argument I had missed either way, thank you for your reply!

closed 2023-04-14 by sourceperl