sourceperl/pyModbusTCP #68: Sending Float Values from Server with Client

AiWsChan, 2023-04-12:
Hi all,

I am new to Github and pyModbusTCP so I hope I post in the right place. 

Currently, I am working on sending value with decimal points (float) from Server to Client. Below is my simplified code

```
def Server():
    print (f"[+]Info: Server is Online! \nIP: {server_ip_address}\nPORT: {server_port}")
    server.start()

    x = 12.25

    x_encode = utils.encode_ieee((x))
    print(f"X_Encoded is {x_encode}")
    
    server.data_bank.set_holding_registers(0, [x_encode])
       
def Client():
    print(f"[+]Info: Server Connected! \nIP: {server_ip_address}\nPORT: {server_port}")
    client.open()
   
    x_reg = client.read_holding_registers(0)
    print(f"X is: {x_reg}")

    x_decode = utils.decode_ieee(x_reg)
    
    print(f"X_decoded is: {x_decode}")

```
**Results:**
X_Encoded is 1102682522
X is [39322]
struct.error: required argument is not an integer (This is for line x_decode = utils.decode_ieee(x_reg))

The question is why when the encoded value (1102682522) goes into register, it changed to another value(39322), what does this means?

Next, since the value is 39322 _should_ be an integer, why does the struct.error message pop up?

Thank you!

`

sourceperl, 2023-04-14:
Hi,

Modbus uses 16-bit words, not 32-bit elements.

So, try this :
```python
from pyModbusTCP import utils
from pyModbusTCP.server import ModbusServer
from pyModbusTCP.client import ModbusClient

# init
server = ModbusServer(no_block=True)
server.start()
client = ModbusClient()
client.open()

# write float to server
srv_float = 12.25
srv_f_32b = utils.encode_ieee(srv_float)
srv_f_32b_msb, srv_f32b_lsb = utils.long_list_to_word([srv_f_32b])
server.data_bank.set_holding_registers(0, [srv_f_32b_msb, srv_f32b_lsb])

# read float with client
cli_f_32b_msb, cli_f32b_lsb = client.read_holding_registers(0, 2)
cli_raw_32b = utils.word_list_to_long([cli_f_32b_msb, cli_f32b_lsb])[0]
cli_float = utils.decode_ieee(cli_raw_32b)
print(f'{cli_float=}')
```

AiWsChan, 2023-04-18:
Hi,

Thank you, thank you so much sir, it works!!

closed 2023-04-18 by sourceperl