sourceperl/pyModbusTCP #69: Write multiple registers with float number: TypeError: 'int' object is not callable

KamRosEmer, 2023-06-28:
Hi, I have a problem with write multiple registers function. 
Code like that: 
```
    ml = ModbusLibrary()
    ml.modbus_init('localhost', 502)
    ml.modbus.open()
    values= [500.5]
    b32_l = [encode_ieee(f) for f in values]
    b16_l = long_list_to_word(b32_l)

    va2 = ml.modbus.write_multiple_registers(0, b16_l)
    print(va2)
    print(str(ml.modbus.last_error()))
    ml.modbus.close()
```

Throws:

`TypeError: 'int' object is not callable`

__init__ functions: 
```
    def __init__(self):
        self.modbus = ModbusClient()
        
    def modbus_init(self, host, port):
        self.modbus.host = host
        self.modbus.port = port
```

What can be wrong?

sourceperl, 2023-06-30:
Hi,

Typically, this type of exception can be raised by misuse of a variable, like here:
```python
a = 42
# raise TypeError: 'int' object is not callable
a()
```

But without the full code and the except traceback, it's really hard to find what's wrong.

An example of python exception traceback, with a reference to the line of code that triggers it:
```
Traceback (most recent call last):
  File "mycode.py", line 6, in <module>
    raise TypeError("Oups!")
TypeError: Oups!
```

Could you share your code and the full except traceback in a github gist for example ?  This really increase your chance to have some helpfull advice here.

Have a good day.

KamRosEmer, 2023-07-05:
Hey, my bad. Looks like modbus.last_error is now a property and I tried to call it with brackets. `ml.modbus.last_error()`

closed 2023-07-05 by KamRosEmer