sourceperl/pyModbusTCP #70: MBAP checking error

FrancoTampieri, 2023-08-10:
Hi I have this sample code:

pyModbusTCP == 0.2.0 python 3.9.2

```
from pyModbusTCP.client import ModbusClient
c = ModbusClient(host="192.168.1.134", port=1024, unit_id=10, debug=True)

c.open()

if c.is_open == False:
    print('not connected')
else:
    print('connected')

print(c.unit_id)

c.read_input_registers(1,1)

c.close()
```

when I try to connect and read the input register 1 (300001) I get this error:

```
connected
10
Tx
[D6 7F 00 00 00 06 0A] 04 00 01 00 01
Rx
[57 65 6C 63 6F 6D 65]
MBAP checking error
```

I used a GUI modbus client and I read the register correctly is 3, why I got this error? Someone can help me? Because of course if I try to print the register I get a None response that is not true.

sourceperl, 2023-08-12:
Hi, 

In the MBAP header, bytes 2-3 are "protocol identifier" and it must be equal to 0 for modbus/TCP. In your debug trace, server return 6C-63 at this place.

Also, if your convert your RAW MBAP data to string it give you 'Welcome'  string:
```python
# return 'Welcome'
print(b'\x57\x65\x6c\x63\x6f\x6d\x65'.decode())
```

Are you sure you connect to a modbus TCP server socket ?

more at: https://en.wikipedia.org/wiki/Modbus#Modbus_TCP_frame_format

FrancoTampieri, 2023-08-12:
Hi, the producer of the machine told me this... of course I think is a custom modbus server and not developed to be full protocol compliant.
I tried with nodes and the modus-serial libs that support both rtu and tcp server and I get the integer value of the register that is 3.
Thank you to giving me the frame format. Is possible to create a custom frame format to handle this case?
The only things that I know is the that the server is a .net server on windows 7
I can use Wireshark for that, today or tomorrow I will try to get all the communications frame.

sourceperl, 2023-08-13:
There is no way to deal with badly formatted MBAP header in pyModbusTCP. You can only create a custom PDU frame (part after MBAP header) with custom_request method, but that's useless here.

Perhaps you can try to change pyModbusTCP code for create a custom version to deal with this specific server but it can be very time-consuming. Another way could be to write a python TCP proxy  for remove/reformat data in front of client.

FrancoTampieri, 2023-08-13:
Yeah I solved developing a program with nodes 18 and the modus-serial libs.

But I will share the tests that I made, because can be useful for someone that face of the same problem with some custom modbus server that was developed without follow the standards :/

Python is the standard of Debian 11 and is versione 3.9.2

I activated the debug for both libs:

```
from pyModbusTCP.client import ModbusClient
c = ModbusClient(host="192.168.1.134", port=1024, unit_id=10, debug=True)

c.open()

if c.is_open == False:
    print('not connected')
else:
    print('connected')

print(c.unit_id)

c.read_input_registers(1,1)

c.close()

from pymodbus.client import ModbusTcpClient
from pymodbus.transaction import ModbusSocketFramer as ModbusFramer

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

ip = "192.168.1.134"
port_num = 1024
unit_num = 10

client = ModbusTcpClient(ip, port=port_num, framer=ModbusFramer)
client.connect()
rr = client.read_input_registers(1,1, unit=unit_num)
print(rr)
```

The output of the two libs are that:
```
connected
10
Tx
[DD 02 00 00 00 06 0A] 04 00 01 00 01
Rx
[57 65 6C 63 6F 6D 65]
MBAP checking error

DEBUG:pymodbus.logging:Connection to Modbus server established. Socket ('192.168.1.125', 46270)
DEBUG:pymodbus.logging:Current transaction state - IDLE
DEBUG:pymodbus.logging:Running transaction 1
DEBUG:pymodbus.logging:SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x4 0x0 0x1 0x0 0x1
DEBUG:pymodbus.logging:New Transaction state "SENDING"
DEBUG:pymodbus.logging:Changing transaction state from "SENDING" to "WAITING FOR REPLY"
DEBUG:pymodbus.logging:Incomplete message received, Expected 28531 bytes Received 19 bytes !!!!
DEBUG:pymodbus.logging:Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
DEBUG:pymodbus.logging:RECV: 0x57 0x65 0x6c 0x63 0x6f 0x6d 0x65 0x20 0x74 0x6f 0x20 0x54 0x63 0x70 0x53 0x72 0x76 0xd 0xa
DEBUG:pymodbus.logging:Processing: 0x57 0x65 0x6c 0x63 0x6f 0x6d 0x65 0x20 0x74 0x6f 0x20 0x54 0x63 0x70 0x53 0x72 0x76 0xd 0xa
DEBUG:pymodbus.logging:Frame check failed, ignoring!!
DEBUG:pymodbus.logging:Resetting frame - Current Frame in buffer - 0x57 0x65 0x6c 0x63 0x6f 0x6d 0x65 0x20 0x74 0x6f 0x20 0x54 0x63 0x70 0x53 0x72 0x76 0xd 0xa
DEBUG:pymodbus.logging:Getting transaction 1
DEBUG:pymodbus.logging:Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response
```

[... 113 lines cut ...]
      0,  0, 0, 0, 0
    ]
    [
      3, 53, 2, 0, 0,
      0,  0, 0, 0, 0
    ]
    [
      3, 53, 2, 0, 0,
      0,  0, 0, 0, 0
    ]
    [
      3, 53, 2, 0, 0,
      0,  0, 0, 0, 0
    ]
    [
      3, 53, 2, 0, 0,
      0,  0, 0, 0, 0
    ]
    [
      3, 53, 2, 0, 0,
      0,  0, 0, 0, 0
    ]
    ^C

Probably the nodes lib don't make some checks on the server and get the data rightly but probably is out of modbus standard standard.

closed 2023-08-13 by FrancoTampieri