sourceperl/pyModbusTCP #73: Modbus receive errors

mattyoungredaptive, 2023-10-27:
We're using a v5.1 LTS Linux kernel, but I've also seen this on Windows dev machine, and Python 3.10.9. I setup the Modbus client with `modbus = ModbusClient(host=self.VVIP, port=502, unit_id=1, auto_open=True, auto_close = False)` and call `modbus.read_holding_registers(0, 20)` once a minute. This works great most of the time. With no changes to the network, I'll very rarely get a receive error, meaning `modbus.read_holding_registers(0, 20)` doesn't return a register list. This will continue for every minutely call, sometimes for a few minutes, sometimes for hours. Then, as suddenly as the read failures started, they stop and everything returns to normal. Closing and re-opening the port has no effect. I haven't been able to find a way to reproduce. Any suggestions?

mattyoungredaptive, 2023-11-08:
After further testing, the exception that I'm hitting is line 658 of client.py: `except socket.error:`

sourceperl, 2023-11-18:
Maybe you can use pyModbusTCP debug mode to see what's wrong here. 

More at: https://pymodbustcp.readthedocs.io/en/latest/quickstart/index.html#modbusclient-debug-mode

sourceperl, 2023-11-28:
Debug mode does not change anything about frame processing. It simply print a frame dump and message (like "connection refused") if an error occur.

mattyoungredaptive, 2023-12-26:
Found the bug, not related to pyModbusTCP

closed 2023-12-26 by mattyoungredaptive