sourceperl/pyModbusTCP #74: Error setting host/port

unixhead, 2023-11-14:
Hi, very useful library! Much appreciate you releasing it, great for learning modbus with. 

I'm finding I get typeerror when setting host/port via the functions with the current release. Works fine if setting host/port via the constructor. Works OK on an older version of pyModbusTCP that I had lying around from a year or two back. 

e.g.
```
from pyModbusTCP.client import ModbusClient
mb = ModbusClient()
mb.host("127.0.0.1")
```
Gives:

```
Traceback (most recent call last):
  File "/home/unixhead/test.py", line 3, in <module>
    mb.host("127.0.0.1")
TypeError: 'str' object is not callable
```

sourceperl, 2023-11-18:
Since version 0.2.0 of pyModbusTCP, "host" parameter is a property.

So to set it:
```python
from pyModbusTCP.client import ModbusClient
mb = ModbusClient()
mb.host = '127.0.0.1'
```

more at: https://github.com/sourceperl/pyModbusTCP/releases/tag/v0.2.0

unixhead, 2023-11-19:
Ah grand thank you, I will get updating my code !

closed 2023-11-19 by unixhead