sourceperl/pyModbusTCP #81: Errno 13 Permission denied

Hipowerguy, 2024-04-22:
Hi,

I am a first time user of pyModbusTCP.

Want to run as a server.  I followed the instructions and it was promissing until I run the code in the "Thonny" Python editor found on my new Raspberry pi 5.

The code begins running properly (the introduction print) but hits "Errno 13 Permission denied" in the server.py, line 1267 in start.  See the details below.

Thanks

Marco
----------------------------
>>> %Run Comm.py
Welcome to Hi-Power Solutions Inc. Modbus communication test program
Implement a Modbus server to answer the PLC requests
Traceback (most recent call last):
  File "/home/marco/pyModbusTCP/pyModbusTCP/server.py", line 1267, in start
    self._service.server_bind()
  File "/usr/lib/python3.11/socketserver.py", line 472, in server_bind
    self.socket.bind(self.server_address)
PermissionError: [Errno 13] Permission denied

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/marco/pyModbusTCP/Comm.py", line 36, in <module>
    server.start()
  File "/home/marco/pyModbusTCP/pyModbusTCP/server.py", line 1270, in start
    raise ModbusServer.NetworkError(e)
pyModbusTCP.server.ModbusServer.NetworkError: [Errno 13] Permission denied

-------------------

[Comm.zip](https://github.com/sourceperl/pyModbusTCP/files/15063417/Comm.zip)
[Result.txt](https://github.com/sourceperl/pyModbusTCP/files/15063422/Result.txt)

sourceperl, 2024-04-22:
Hello, in Linux port numbers below 1024 cannot be used by a process running by a normal user. To fix this, you can run your script as root or open another port like 5020 (instead of the standard modbus port tcp/502).

Hipowerguy, 2024-04-22:
Thanks for the quick response.
It solved the crashing issue.  

Now the program flow seems to stop after the 'server.start' is issued.  No other code runs afterward.
How to get the data out if the program flow is interrupted?

Another question.

Is this server running as an independant process with the results available to others?
The objective is to use the results from the Modbus communication to fed another application.

Thanks,

sourceperl, 2024-04-22:
- See https://github.com/sourceperl/pyModbusTCP/blob/master/examples/server_schedule.py for a non-blocking server example.
- Regarding your second question, if you enable no_block mode, the server is run as an independent thread. 
- To deal with modbus server read/write actions you can look example at https://github.com/sourceperl/pyModbusTCP/blob/master/examples/server_virtual_data.py.

Hipowerguy, 2024-04-23:
Hi,

I am still trying to set my Raspberry pi as a modbus TCP server.
My computer is the client reading the first few holding registers loaded with values as shown below.

While the program exits (running in Thonny), the result is the same wether I enable or desable the block in the server.py file.

Question: 
1-Since I never stop the server, is there an issue with multiple starts in different tests?
2-Is there an issue with running ModbusServer using Thonny?

Thanks.

Code sample:
# start modbus server
print('start modbus server')
server = ModbusServer(host=args.host, port=args.port)

#add data to read
server.data_bank.set_holding_registers(0, [0,11,22,33,44,55,66,77])
server.start()
print('Started, end program')

sourceperl, 2024-04-25:
Hi,

> Since I never stop the server, is there an issue with multiple starts in different tests?

I'm not sure I understand this correctly, but you can't start two server instances that bind to the same TCP port at the same time.

> Is there an issue with running ModbusServer using Thonny?

I don't know any problem with Thonny.

### an example that may help : 
```python
import time
from pyModbusTCP.server import ModbusServer

# listen on TCP port 5020 for all server IP addresses
server = ModbusServer(host='0.0.0.0', port=5020, no_block=True)
server.start()

# this code will be run only if "no_block=True" is set in ModbusServer
print('server is start')
# an infinite loop allows the server to remain active
# otherwise the script ends and therefore the server thread too
while True:
    for my_counter in range(0, 1000):
        server.data_bank.set_holding_registers(0, [my_counter])
        time.sleep(1.0)
```

closed 2024-07-30 by sourceperl