sourceperl/pyModbusTCP #88: Unable to import from pyModbusTCP

Style177, 2024-11-04:
Hi everybody out there. First of all I must say that I am very new to linux systems. I am running debian on a Raspberrypi and managed to create a dashboard to visualize my Huawei pv-plant (SUN2000 via evcc/TCP - influxdb - grafana). Now I wanted to add some data from the SUN2000 Dongle that are not published by evcc.

For this purpose, in the first step I was looking for a simple utility to read registers and found pyModbusTCP.   Unfortunately I failed at this very first step. Now, after some hours of trying, I hope to find some help here. I fear that there is something very simple that I am missing. Maybe I can't see something that is right before my eyes.

For your info:
python3.11 is up and running, pyModbusTCP is installed (or seems to be so): 
```
(myenv) emilio@raspberrypi:~ $ python3 -m pip install pyModbusTCP
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pyModbusTCP in ./myenv/lib/python3.11/site-packages (0.3.0)
```
However, a module pyModbusTCP.client cannot be found:
```
(myenv) emilio@raspberrypi:~ $ python3
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyModbusTCP.server import ModbusServer
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/emilio/pyModbusTCP.py", line 1, in <module>
    from pyModbusTCP.client import ModbusClient
ModuleNotFoundError: No module named 'pyModbusTCP.client'; 'pyModbusTCP' is not a package
>>> 
```
further info can be supplied, if necessary.

Emilio

sourceperl, 2024-11-05:
Hello, it's probably because you call your script pyModbusTCP.py (same name as the module). Just call it myapp.py for example and it should be fine.

Style177, 2024-11-05:
Wow, thank you so much! Indeed, there was confusion in the directory (probably the result of too much nightwork!). It contained a) a subdirectory "pyModbusTCP", b) a Python script "test_pyModbusTCP.py" and c) another Python script "pyModbusTCP". So, scipt b) referenced script c) instead of directory a).
So, as I assumed, it was right before my eyes, but I was unable to see it.

closed 2024-11-05 by Style177