sourceperl/pyModbusTCP #89: with loglevel set to DEBUG, TypeError: hexlify() takes no keyword arguments

thomasgoedertier, 2024-11-26:
Hi,

When I set the loglevel for the pyModbusTCP.client to logging.DEBUG
I get following error when trying to write a register

```
Traceback (most recent call last):
  File "/user/user.py", line 162, in <module>
    modbusResponse = send_modbus_request(modbus_client, register, payload)
  File "/user/user.py", line 99, in send_modbus_request
    modbus_response = client.write_single_register(register_address,payload)
  File "/usr/lib/python3.7/site-packages/pyModbusTCP/client.py", line 606, in write_single_register
    rx_pdu = self._req_pdu(tx_pdu=tx_pdu, rx_min_len=5)
  File "/usr/lib/python3.7/site-packages/pyModbusTCP/client.py", line 910, in _req_pdu
    self._send_pdu(tx_pdu)
  File "/usr/lib/python3.7/site-packages/pyModbusTCP/client.py", line 800, in _send_pdu
    self._on_tx_rx(frame=tx_frame, is_tx=True)
  File "/usr/lib/python3.7/site-packages/pyModbusTCP/client.py", line 938, in _on_tx_rx
    mbap_s = hexlify(frame[0:7], sep=' ').upper().decode()
TypeError: hexlify() takes no keyword arguments
```

https://github.com/sourceperl/pyModbusTCP/blob/809cf98df9b526b19416f1e043a215e7168efd73/pyModbusTCP/client.py#L938

putting loglevel to logging.INFO or higher mitigates the issue

Kind regards

thomasgoedertier, 2024-11-26:
version installed is latest 0.3.0
root@device:~# pip3 show pyModbusTCP
Name: pyModbusTCP
Version: 0.3.0
Summary: A simple Modbus/TCP library for Python
Home-page: https://github.com/sourceperl/pyModbusTCP
Author: Loic Lefebvre
Author-email: loic.celine@free.fr
License: MIT
Location: /usr/lib/python3.7/site-packages
Requires:
Required-by:

sourceperl, 2024-11-30:
Hello, pyModbusTCP is not supported on python 3.7 (which is deprecated since June 2023).

closed 2024-11-30 by sourceperl