sourceperl/pyModbusTCP #91: is_open doesn't work

VictoriaPresnyakova, 2024-12-06:
Hi!
is_open doesn't return the correct status when I check connection without trying to read or write

AnderssonDavid, 2025-01-14:
I have noticed this as well. I just poll a random register within a "try/except" block. After a poll, the "is_open" does work for me.

sourceperl, 2025-01-15:
Hi, in the ModbusClient code, the is_open property depends on the state of the Python socket like this :

```python
    @property
    def is_open(self):
        """Get current status of the TCP connection (True = open)."""
        return self._sock.fileno() > 0
```

From my testing, the socket state is updated only when we call a send() or recv() on that socket. So yes, you need to try a read/write to update the socket state.

charliehuddleston, 2025-01-15:
I just verified the following works, is this how to reproduce the issue?

code1.py or in a terminal window:
```python
from pyModbusTCP.server import ModbusServer, DataBank
mbdb = DataBank()
s = ModbusServer(host="0.0.0.0", port=2502, data_bank=mbdb)
s.start()
```

code2.py or in a second terminal window:
```python
from pyModbusTCP.client import ModbusClient
c = ModbusClient(host="127.0.0.1", port=2502, auto_open=False, auto_close=False)
c.open()
```
c.is_open shows True without performing any read/write operations.