sourceperl/pyModbusTCP #92: Issue(??) with accessing R/W registers

DrUpcon, 2025-02-03:
Hi,

_just as info ahead, I'm also in contact with the device manufacturer, but they are not the most cooperative ones because I don't want to use their weird software package._

So, I have a modbus TCP device which acts as an temperature controller, running in slave mode. I can access and connect to it fairly nicely with both, your library as well as their software.
The weird thing I don't understand:

- I have the adress for a control value (here: 0x670A), according to the modbus documentation of the manufacturer it is always in R/W mode. I can read the value there using your library, but I can't write to it. From debugging, I get this error:
`pyModbusTCP.client - DEBUG - (192.***.***.***:255) modbus exception (code 2 "modbus exception")`
- Some further, but not too accurate debugging revealed, I get randomly specific Modbus errors when using your library
1. Exception Response 131 / 0: "Illegal Address" (this one is by far the most)
2.  Exception Response 144 / 0: "Illegal Function"

The illegal address is what confuses me, I use the exact same address as for reading the data and this works just fine.

`
from pyModbusTCP.client import ModbusClient
from icecream import ic
import struct
import logging
ipaddr="192.***.***.***" # blanked for use in public space
addr=26378 # decimal for 0x670A
port=502
def write_temp(addr, value=350.0):
    # unit_id needs to be 255 in slave mode
    client = ModbusClient(host=ipaddr, port=port, unit_id=255, auto_open=True)
    try:
        # Convert float to CDAB MSB
        modbus_temp = float_to_modbus_cd_ab(value)
        
        # write to register
        response = client.write_single_register(addr, modbus_temp)
        
        if response:
            print(f"successfull written: {modbus_temp}")
        else:
            print("Error on writing temperature.")
    except Exception as e:
        print(f"Error: {e}")
# closing happens at a later point in the program`

I can see on the display of the controller, that communication is happening, but it is not taking the value I give it to the register.
Now the part where it gets really confusing to me. The controller has the function the create readable or writeable "frames", basically multiple registers attached to each other but with a different address. The whole purpose is to reduce data-transfer when moving a lot of data frequently, which I'm not planning to do.

If I now use the same code as shown above just with the adress 0x8000 (32768 in decimal), I don't get an error message but something else, very weird is happening. The value I have to provide is again the weird modbus 32bit float following IEEE754, so with the bits swapped to CDAB, but I'm not sure what is happening in process.
1. "A" number is being written to the frame
2. The number is not interpreted correctly, my conversion is correct, the data type is correct, but still, not in CDAB, but also in no other bit swap configuration I get the correct number written into the frame. I know it should be possible, because I can do this using the software of the manufacturer (directly into 0x670A as well as into 0x8000)
E.g.: I enter 25.0 for 25.0°C, it will convert according to the IEEE to 17326, but shows up as -1.37e-38 °C in the device (which is not even the fallback error value)
4. If I submit an INT value directly, it will show up in the exact value I've entered (but can't be used, since it's an int and it requires a float, trust me...I tried)

So....as you can guess, I'm heck of confused by now. I didn't posted the device manufacturers name here on purpose, don't wanna blame anyone. So maybe you've an idea if this is really a bug or not.
I'm pretty sure the modbus part from the device is alright, their software uses the protocol as well and works just fine. But it's propriertary and does not allow any communication with other programs, which is a big issue for me

Best regards
the doctor

sourceperl, 2025-02-14:
Hello, if I try to summarize quickly, you are talking about writing something on 32 bits. But you seem to be using function 6 "Write Single Holding Register" which only transfer a 16 bits word. This may be the origin of the problem.

To keep thing simple, just try this from python REPL:
```python
from pyModbusTCP.client import ModbusClient
import struct

c = ModbusClient(host=ipaddr, port=port, unit_id=255)
# convert float to 32 bits IEEE
set_point = 25.0
ieee_bytes = struct.pack('<f', set_point)
# truncate 32 bits bytes to 2x 16bits words
up_16bits, low_16bits = struct.unpack('<HH', ieee_bytes)
# write 2 words at address 0x670A
c.write_multiple_registers(0x670A, [up_16bits, low_16bits])
# to find the correct byte order, you can also try this (with or without modify the unpack encoding)
c.write_multiple_registers(0x670A, [low_16bits, up_16bits])
```

I hope this helps you.

DrUpcon, 2025-02-14:
Hi,

sorry, I forgot to reply earlier. I was able to solve the issue last friday. Previously I reached out to the manufacturer and mailed the firmware of the controller back to them because of some bugs in the display. The firmware was found to have several flaws from factory, also on the modbus part. Beside wrong addresses in the official documentation, the main issue was that I can't write in the "R/W" labelled registers directly, instead I have to use a specific "input" register which than internally transfers the value to the R/W labelled variable. In the current modbus documentation of the manufacturer, this is simply missing.

No clue why....but after I got the updated registers and structures I was able to get the code working.

Sorry about the bit confusion, when I modified my code to a MWE, I somehow dropped the splitting into 16bit. But earlier this week I also switched to an approach that uses the write_multiple_registers approach.

Anyway, thank you for your package and the help here! I really appreciate your work here. 

this is my current (working) method:
'''
def write_temp(addr, value=3.0):
    # change to (CDAB MSB)                                         
    modbus_temp1, modbus_temp2= float_to_modbus(value)

    # Checking if list
    if isinstance(modbus_temp, int):
        modbus_temp = [modbus_temp2, modbus_temp1]

    # Write registers
    response = client.write_multiple_registers(addr, modbus_temp)

    if response:
        print(f"Written: {value} °C successfully.")
    else:
        print("Error while writing.")
'''

closed 2025-02-14 by sourceperl